WebODM/WebODM #1477: Possible Bug: Cloud Import / DroneDB Import Buttons Break UI

Saijin-Naib, 2024-03-19:
### How did you install WebODM (docker, installer, etc.)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

N/A

### What is the problem?

When using the WebUI and clicking on the Cloud Import or DroneDB Import buttons, the whole UI below the top bar disappears.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

All current/latest git/docker, currently affected are Cloud Import and DroneDB Import buttons on Task. Click them and the page goes empty.

```
03:35:47.968 Uncaught Error: Minified React error #188; visit https://reactjs.org/docs/error-decoder.html?invariant=188 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    findDOMNode http://localhost:8000/plugins/cloudimport/build/ImportView.js:2
    default http://localhost:8000/plugins/cloudimport/build/ImportView.js:2
    addEventListeners http://localhost:8000/plugins/cloudimport/build/ImportView.js:2
    componentDidUpdate http://localhost:8000/plugins/cloudimport/build/ImportView.js:2
    cs http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    Ll http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    unstable_runWithPriority http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    Ko http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    Sl http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    pl http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    Yo http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    unstable_runWithPriority http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    Ko http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    Yo http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    $o http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    N http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
    Qe http://localhost:8000/static/app/bundles/main-b7228f828b649d970e13.js:2
```
[image]

Potential issue with Bootstrap or something else?

Saijin-Naib, 2024-05-20:
Issue confirmed on MacOS+Docker:  
https://community.opendronemap.org/t/dronedb-import/20401/4?u=saijin_naib

pierotofy, 2024-09-29:
I cannot reproduce this with the latest version, so I assume this has been fixed? 

If the problem persists, feel free to re-open. :pray:

closed 2024-09-29 by pierotofy