WebODM/WebODM #1479: error message processing oblique data

Rico195786, 2024-03-20:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************
### How did you install WebODM (docker, installer, etc.)?

installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Edge 122 on Windows 11]

### What is the problem?

I get an error processing oblique data. The error makes reference to me using the best practices of data acquisition. I was able to process the data using Pix4D, Terra, and Metashape, so I am guessing my practices were correct. If you visit my website, www.thundercitydrone.com and go to the 3D Digital Image page you will see the data processed with Pix4D. 

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

I'm using the default parameters for processing. 

I uploaded it to google drive last time and I asked what email I use to give access and I never heard back, so I'm confused. Is there a different google drive? Is someone really going to help?  

[image]

closed 2024-03-20 by github-actions[bot]

Rico195786, 2024-03-20:
I need an email to give access to my google drive

pierotofy, 2024-03-21:
Can you post a link to it?