WebODM/WebODM #1482: error processing oblique data that processed ok on 3 other software's. visit thundercitydrone.com and go to the 3D Digital Twin page to see the result of processing the data on Pix4D.

Rico195786, 2024-03-21:
### How did you install WebODM (docker, installer, etc.)?

[installer]

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Edge 122 on Windows 11]

### What is the problem?

[stops processing with error. see attached screenshots.]

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

[process the attached data.

I used the default settings.

[image]
[image]
[image]

https://drive.google.com/open?id=1NQLTNc1C5XjZovaQeirQPL-ZtSQpYuN6&usp=drive_fs]

Rico195786, 2024-04-21:
Has there been an update to this? Is it a software issue or something I'm doing wrong?  Thanks

smathermather, 2024-07-21:
Uncertain, but the better place to troubleshoot this is at community.opendronemap.org. I'll close this issue for now, and we can re-open here if it is in-fact a bug. Thanks!

closed 2024-07-21 by smathermather