WebODM/WebODM #1500: Quntum Drones D2M Images can be processed

globalmatrix, 2024-05-09:
Currently we are using a Quatum Drone with a 5 lense camera D2M. Running the latest vertion of WebODM with my Ubuntu linux AMD Ryzen Threadripper 3960X 24-Core Processor, 48 and 125 GB of RAM we get that the dataset can be processed. We have 75% lateral and 75% forward overlap. Our Sensefly ebee with SODA camera works perfectly. Any suggestion? if anyone is interested please let me know to share dataset.

pierotofy, 2024-05-10:
If you have issues processing a dataset, make sure to include a link to the images, as indicated :pray: 

>  If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive

globalmatrix, 2024-05-14:
Hi Piero:
I the following link i am sharing a nice data set of D2M Quantum Drone 5 lense camera .
https://drive.google.com/drive/folders/1Vh9XSiEJdsZDuBR24DSsum6SBVhOORDT?usp=drive_link

pierotofy, 2025-03-12:
I finally had time to try this dataset; interestingly, I think the software struggles to group together the 5 sensors because they are not treated as a rig.

Subset: https://cloud1.webodm.net/public/task/54829362-abbb-4017-9730-0dafb1eb4143/3d/

My best workaround for working with this sensor at the moment would be to discard all non-nadir images.

In the future we could potentially explore adding rig support to ODM.