WebODM/WebODM #1528: Duplicating task prior to processing results in tasks that are entangled

smathermather, 2024-07-21:
### How did you install WebODM (docker, installer, etc.)?

Docker deploy on Ubuntu 24.04

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Brave, Chromium, Firefox

### What is the problem?

If one uploads images to a new task, processes the data, and the duplicates the task, everything works as expected.

If one uploads images to a new task, cancels the task before finishing processing, and then duplicates the canceled task, WebODM seems to confuse the two tasks, for example, restarting each task will result in seeing the same logging showing in each task.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

If one uploads images to a new task, cancels the task before finishing processing, and then duplicates the canceled task, WebODM seems to confuse the two tasks, for example, restarting each task will result in seeing the same logging showing in each task.

Happens with any dataset, but to keep the bot happy, here's a link:
link.com

lfeldb, 2024-08-13:
I can confirm this issue.
Also it does not only seem to happen on canceling the task manually, but also when tasks are completed (succesfully with an error) and then copied.

I also just noticed when i create a copy, the field 'name' is not updated in /var/www/data/tasks.json in nodeodm docker container. It seems that this file is created not at start-time, but later on in the process.

What does the 'off-by-one error' tag mean? Can I assist with some way?

lfeldb, 2024-08-13:
OK, I'll look into it as soon as I have some time. Thank you for the pull request reference, this saves me some time getting familiar with the code.

( Alright! I thought off-by-one might have suggested an indexing error, or it was one fix away from being fixed :-)

smathermather, 2024-08-13:
> OK, I'll look into it as soon as I have some time. Thank you for the pull request reference, this saves me some time getting familiar with the code.

Awesome. Thanks!

> ( Alright! I thought off-by-one might have suggested an indexing error, or it was one fix away from being fixed :-)

It was a tag I added a long time ago as a joke, and sometimes it is at least almost apropos. :slightly_smiling_face:

pierotofy, 2024-09-28:
Should be fixed with #1559

closed 2024-09-28 by pierotofy