WebODM/WebODM #1532: Permanent shapefile/geojson layer

fmgltdlv, 2024-07-31:
I apologize in advance as I know the import temporary shapefile/geojson feature already exists. I have a single shapefile that spans an entire city for a utility network that I want to be able to overlay in any project I open. For me, I don't mind dragging and dropping the file everytime or opening the shapefiles and orthos in google earth or qgis, but I also share these projects with a team of people who also use webodm on their phone and explaining to some of them how to do this isn't practical. 

Is there someway to make it so that I can embed a permanent layer into the file structure that WebODM can always reference?

pierotofy, 2024-07-31:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

fmgltdlv, 2024-07-31:
> Could we move this conversation over to the forum at https://community.opendronemap.org? 🙏 The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! 👍

Hello. It was a feature I was trying to request. I've tried asking and searching on the forum and it doesn't appear this feature exists.

pierotofy, 2024-07-31:
Ah, ok, we can keep it open then. :pray:

pierotofy, 2024-08-08:
Does Annotate Pro solve your problem? If so, you could consider purchasing it.

closed 2024-09-28 by pierotofy