WebODM/WebODM #1566: [ERROR] Directory submodels does not exist

maphew, 2024-10-23:
### How did you install WebODM (docker, installer, etc.)?

Windows installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Edge 130 on Windows 10

### What is the problem?

Processing ~2500 images fails to run after 6 hours (using fast-ortho, split:1250). Log file ends with _"[ERROR]   Directory submodels does not exist."_. Running the command line immediately before this error manually from command prompt yields same message:
```
C:\>"C:\WebODM\resources\app\apps\ODM\run" ^
More?   --feature-type hahog --fast-orthophoto --auto-boundary --orthophoto-resolution 2.0 --gltf --orthophoto-cutline ^
More?   --dem-euclidean-map --skip-3dmodel --skip-report --project-path ^
More?   "C:\WebODM\resources\app\apps\NodeODM\data\443f1988-6197-4247-b31f-2f644fd5679a\submodels" submodels
[INFO]    Fast orthophoto is turned on, automatically setting --skip-3dmodel
[INFO]    Initializing ODM 3.5.1 - Wed Oct 23 08:46:13  2024
[ERROR]   Directory submodels does not exist.
C:\>
```
Looking higher up in the WebODM logfile we see that the submodels do exist, but at a different path (emphasis added):

> [INFO]    C:\WebODM\resources\app\apps\NodeODM\data\443f1988-6197-4247-b31f-2f644fd5679a\opensfm\submodels\submodel_0031\reconstruction.aligned.json is now C:\WebODM\resources\app\apps\NodeODM\data\443f1988-6197-4247-b31f-2f644fd5679a\opensfm\submodels\submodel_0031\reconstruction.json

They're at `2f644fd5679a\opensfm\submodels` not `2f644fd5679a\submodels` of the command line.

Extended log tail:
```
[INFO]    C:\WebODM\resources\app\apps\NodeODM\data\443f1988-6197-4247-b31f-2f644fd5679a\opensfm\submodels\submodel_0030\reconstruction.aligned.json is now C:\WebODM\resources\app\apps\NodeODM\data\443f1988-6197-4247-b31f-2f644fd5679a\opensfm\submodels\submodel_0030\reconstruction.json
[INFO]    C:\WebODM\resources\app\apps\NodeODM\data\443f1988-6197-4247-b31f-2f644fd5679a\opensfm\submodels\submodel_0031\reconstruction.aligned.json is now C:\WebODM\resources\app\apps\NodeODM\data\443f1988-6197-4247-b31f-2f644fd5679a\opensfm\submodels\submodel_0031\reconstruction.json
[INFO]    ========================
[INFO]    Processing submodels
[INFO]    ========================
[INFO]    running "C:\WebODM\resources\app\apps\ODM\run" --feature-type hahog --fast-orthophoto --auto-boundary --orthophoto-resolution 2.0 --gltf --orthophoto-cutline --dem-euclidean-map --skip-3dmodel --skip-report --project-path "C:\WebODM\resources\app\apps\NodeODM\data\443f1988-6197-4247-b31f-2f644fd5679a\submodels" submodels
[INFO]    Fast orthophoto is turned on, automatically setting --skip-3dmodel
[INFO]    Initializing ODM 3.5.1 - Tue Oct 22 17:50:52  2024
[ERROR]   Directory submodels does not exist.

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
File "C:\WebODM\resources\app\apps\ODM\stages\odm_app.py", line 82, in execute
self.first_stage.run()
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 471, in run
self.next_stage.run(outputs)
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 450, in run
self.process(self.args, outputs)
File "C:\WebODM\resources\app\apps\ODM\stages\splitmerge.py", line 192, in process
system.run(" ".join(map(double_quote, map(str, argv))), env_vars=os.environ.copy())
File "C:\WebODM\resources\app\apps\ODM\opendm\system.py", line 112, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/
```

Some more experimenting at the cmd prompt reveals that running with `--project-path "C:\....\opensfm" submodels" (path simplified) gets a little farther but then complains there are no images. Eventually I realised this would need to be run for each of the submodels in turn, like so `C:\...\opensfm\submodels" submodel_0000`:

```
"C:\WebODM\resources\app\apps\ODM\run" ^
  --feature-type hahog --fast-orthophoto --auto-boundary --orthophoto-resolution 2.0 --gltf --orthophoto-cutline ^
  --dem-euclidean-map --skip-3dmodel --skip-report --project-path ^
  "C:\WebODM\resources\app\apps\NodeODM\data\443f1988-6197-4247-b31f-2f644fd5679a\opensfm\submodels" submodel_0000
```

It took 6 hours to get to this point. Aside from repeating this command for each of the 31 submodel folders, what other commands can I run to manually to finish the job?

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Task settings:

**Images:** 2489 (from DJI Matrice 300, P1 camera, )

Format                                   : JPEG
File size                                : 20.6 MiB
Format                                   : JPEG
Width                                    : 8 192 pixels
Height                                   : 5 460 pixels
Color space                              : YUV
Chroma subsampling                       : 4:4:4
Bit depth                                : 8 bits
Compression mode                         : Lossy
Stream size                              : 20.6 MiB (100%)
ColorSpace_ICC                           : RGB

**Processing Node:** node-odm-1 (manual)
**Options:**  auto-boundary: true, fast-orthophoto: true, feature-type: hahog, orthophoto-resolution: 2, split: 1250, rerun-from: odm_postprocess
**Disk Usage:**  37.93 Gb

Free disk space: 300 GB
Memory: 64 GB (used 16%)

A factor that might be involved: I first ran this job without split but it ran out of memory. For this run I edited the task, added _split_, and ran using _Resume_.

closed 2024-10-23 by github-actions[bot]

jet86, 2024-10-28:
You didn't include a copy of your dataset as required by the [issue guidelines](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), and as noted by the bot.

maphew, 2024-10-30:
I didn’t think it made sense to upload a 38GB dataset (if there’s even a place that accepts that much) without doing some work to narrow down the problem first. At any rate, for future searchers there more info on this in the forum - https://community.opendronemap.org/t/error-directory-submodels-does-not-exist/

smathermather, 2024-11-01:
Thanks for linking this back @maphew