WebODM/WebODM #1567: Cesium Ion Plugin activated prevents loading the task contents of the dashboard in any project

luismgarciac, 2024-10-28:
### How did you install WebODM (docker, installer, etc.)?

Windows 11 Home, WSL, Ubuntu 22.04.5 LTS, Docker Desktop
WebODM 2.5.5

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Edge 130 on Windows 11](https://www.whatismybrowser.com/detect/what-version-of-edge-do-i-have/)
[Chrome 130 on Windows 11](https://www.whatismybrowser.com/detect/what-version-of-chrome-do-i-have/)

### What is the problem?

I am trying to use the Cesium Ion Plugin to automatically upload and georeference a 3D textured model created with WebODM, but when I activate the Cesium and go to the dashboard I am unable to open any dataset from any model, the menu of the dataset goes blank, so I am not even able to try the Cesium Ion Plugin.

However, if I upload the data directly to Cesium Ion, the data is visible but not georeferenced, and has to be georeferenced manually.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

I have uploaded to Google Cloud a dataset that I intend to use for this purpose, but any other dataset produces the same results:
https://drive.google.com/file/d/1TYiYmjVLzHNGAJT1A5i73op-Ft1ZPVQQ/view?usp=drive_link

Before processing I made sure that the texturing-single-material setting was enabled as indicated in the Cesium Ion integration documentation:
https://cesium.com/learn/ion/integrations-learn/integrations-webodm/

When the Cesium Ion plugin is enabled the dashboard content goes blank as one project task is expanded:
[image]

When the Cesium Ion plugin is disabled the dashboard content becomes visible, but it is not possible to use the plugin:
[image]

The developer console provides the following error message:
[image]
`main-513c83c….js:2 
 TypeError: Cannot read properties of undefined (reading 'POINTCLOUD')
    at eval (TaskView.js:2:16907)
    at Array.sort (<anonymous>)
    at r.value (TaskView.js:2:16867)
    at Ua (main-513c83c….js:2:216596)
    at Fa (main-513c83c….js:2:216392)
    at Ts (main-513c83c….js:2:251928)
    at kl (main-513c83c….js:2:243411)
    at xl (main-513c83c….js:2:243336)
    at pl (main-513c83c….js:2:240369)
    at main-513c83c….js:2:192083
console.error	@	main-513c83c….js:2
rs	@	main-513c83c….js:2
n.callback	@	main-513c83c….js:2
_r	@	main-513c83c….js:2
cs	@	main-513c83c….js:2
Ll	@	main-513c83c….js:2
t.7463.e.unstable_runWithPriority	@	main-513c83c….js:2
Ko	@	main-513c83c….js:2
Sl	@	main-513c83c….js:2
pl	@	main-513c83c….js:2
(anonymous)	@	main-513c83c….js:2
t.7463.e.unstable_runWithPriority	@	main-513c83c….js:2
Ko	@	main-513c83c….js:2
Yo	@	main-513c83c….js:2
$o	@	main-513c83c….js:2
ul	@	main-513c83c….js:2
enqueueSetState	@	main-513c83c….js:2
w.setState	@	TaskView.js:2
eval	@	TaskView.js:2
Promise.then		
eval	@	TaskView.js:2
value	@	TaskView.js:2
cs	@	main-513c83c….js:2
Ll	@	main-513c83c….js:2
t.7463.e.unstable_runWithPriority	@	main-513c83c….js:2
Ko	@	main-513c83c….js:2
Sl	@	main-513c83c….js:2
pl	@	main-513c83c….js:2
(anonymous)	@	main-513c83c….js:2
t.7463.e.unstable_runWithPriority	@	main-513c83c….js:2
Ko	@	main-513c83c….js:2
Yo	@	main-513c83c….js:2
$o	@	main-513c83c….js:2
ul	@	main-513c83c….js:2
enqueueSetState	@	main-513c83c….js:2
(anonymous)	@	Dashboard-513c83c….js:2
(anonymous)	@	Dashboard-513c83c….js:2
(anonymous)	@	main-513c83c….js:2
t.__emitToSubscription	@	main-513c83c….js:2
t.emit	@	main-513c83c….js:2
(anonymous)	@	main-513c83c….js:2
setTimeout		
r	@	main-513c83c….js:2
(anonymous)	@	main-513c83c….js:2
Promise.then		
(anonymous)	@	main-513c83c….js:2
t.__emitToSubscription	@	main-513c83c….js:2
t.emit	@	main-513c83c….js:2
(anonymous)	@	main-513c83c….js:2
value	@	Dashboard-513c83c….js:2
cs	@	main-513c83c….js:2
Ll	@	main-513c83c….js:2
t.7463.e.unstable_runWithPriority	@	main-513c83c….js:2
Ko	@	main-513c83c….js:2
Sl	@	main-513c83c….js:2
pl	@	main-513c83c….js:2
(anonymous)	@	main-513c83c….js:2
t.7463.e.unstable_runWithPriority	@	main-513c83c….js:2
Ko	@	main-513c83c….js:2
Yo	@	main-513c83c….js:2
$o	@	main-513c83c….js:2
N	@	main-513c83c….js:2
Qe	@	main-513c83c….js:2`

pierotofy, 2025-02-08:
Thanks for the report. This might be fixed by #1597 (version 2.7.0); if the issue persists, feel free to re-open? 🙏

closed 2025-02-08 by pierotofy

iammjm, 2025-02-27:
Hello, I encounter the same issue (Cesium Ion Plugin activated prevents loading the task contents of the dashboard in any project). 
This makes the plugin completely unusable for me.
Tested on webODM on Win11 and on a separate instance of webODM on Ubuntu via MS Edge. 

dev console error: "TypeError: Cannot read properties of undefined (reading 'POINTCLOUD')
    at eval (TaskView.js:2:16907)
    at Array.sort (<anonymous>)
    at r.value (TaskView.js:2:16867)
    at Ua (main-bd828bc903d8d894af90.js:2:216596)
    at Fa (main-bd828bc903d8d894af90.js:2:216392)
    at Ts (main-bd828bc903d8d894af90.js:2:251928)
    at kl (main-bd828bc903d8d894af90.js:2:243411)
    at xl (main-bd828bc903d8d894af90.js:2:243336)
    at pl (main-bd828bc903d8d894af90.js:2:240369)
    at main-bd828bc903d8d894af90.js:2:192083"