WebODM/WebODM #1603: Runtime `OSError: [Errno 95]` .

MubinMuhammad, 2025-02-14:
After cloning webodm, I've run `./webodm.sh start`. After running, the following error is shown repeatedly:

```bash
webapp      | Traceback (most recent call last):
webapp      |   File "/usr/local/lib/python3.9/dist-packages/gunicorn/sock.py", line 44, in set_options
webapp      |     sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEPORT, 1)
webapp      | OSError: [Errno 95] Operation not supported
webapp      |
webapp      | During handling of the above exception, another exception occurred:
webapp      |
webapp      | Traceback (most recent call last):
webapp      |   File "/usr/local/bin/gunicorn", line 8, in <module>
webapp      |     sys.exit(run())
webapp      |   File "/usr/local/lib/python3.9/dist-packages/gunicorn/app/wsgiapp.py", line 74, in run
webapp      |     WSGIApplication("%(prog)s [OPTIONS] [APP_MODULE]").run()
webapp      |   File "/usr/local/lib/python3.9/dist-packages/gunicorn/app/base.py", line 203, in run
webapp      |     super(Application, self).run()
webapp      |   File "/usr/local/lib/python3.9/dist-packages/gunicorn/app/base.py", line 72, in run
webapp      |     Arbiter(self).run()
webapp      |   File "/usr/local/lib/python3.9/dist-packages/gunicorn/arbiter.py", line 198, in run
webapp      |     self.start()
webapp      |   File "/usr/local/lib/python3.9/dist-packages/gunicorn/arbiter.py", line 157, in start
webapp      |     self.LISTENERS = sock.create_sockets(self.cfg, self.log, fds)
webapp      |   File "/usr/local/lib/python3.9/dist-packages/gunicorn/sock.py", line 180, in create_sockets
webapp      |     sock = sock_type(addr, conf, log)
webapp      |   File "/usr/local/lib/python3.9/dist-packages/gunicorn/sock.py", line 117, in __init__
webapp      |     super(UnixSocket, self).__init__(addr, conf, log, fd=fd)
webapp      |   File "/usr/local/lib/python3.9/dist-packages/gunicorn/sock.py", line 32, in __init__
webapp      |     self.sock = self.set_options(sock, bound=bound)
webapp      |   File "/usr/local/lib/python3.9/dist-packages/gunicorn/sock.py", line 46, in set_options
webapp      |     if err[0] not in (errno.ENOPROTOOPT, errno.EINVAL):
webapp      | TypeError: 'OSError' object is not subscriptable
worker      | /usr/local/lib/python3.9/dist-packages/celery/platforms.py:800: RuntimeWarning: You're running the worker with superuser privileges: this is
worker      | absolutely not recommended!
```

ppetru, 2025-02-15:
I ran into the same issue. I had been updating by just running git pull on my existing clone, and missed the part of the README that explains how to update. After doing ./webodm.sh update, things are working properly again.

pierotofy, 2025-02-15:
This should have been fixed some time ago with https://github.com/OpenDroneMap/WebODM/pull/1587

Make sure to run `./webodm.sh update` as suggested by ppetru? 🙏 

If the issue persists, feel free to reopen.

closed 2025-02-15 by pierotofy

livmion, 2025-03-19:
The issue is still present in Debian Bookworm with backported kernel image 6.12.12+bpo-amd64. The repository was cloned with the last version. Downgrading the kernel to 6.9 solves the problem and WebODM can run correctly.

For the record, Debian Trixie with kernel 6.12.17-amd64 has not the same problem.