WebODM/WebODM #1609: Contour button is not on screen

nbricker19, 2025-02-16:
Contour button does not appear on screen. I looked through the forum and found that I should toggle the plug in enable/disable switch, but that did not work either. Any assistance would be greatly appreciated.

Rgswick, 2025-02-16:
I have the same problem, it was there before I installed the latest update

cpo-ux, 2025-02-17:
Same issue for me, Window 11 install.  Contour button doesn't appear on any of my projects.  This is with Windows installer version 2.7.0.  I installed an older version (2.5.6) and the contour button was back in 2D map view.

pierotofy, 2025-02-17:
This was fixed with https://github.com/OpenDroneMap/WebODM/pull/1604

You'll need to wait until an update for the Windows Installer is released (a few days, probably).

closed 2025-02-17 by pierotofy

Rgswick, 2025-02-17:
Ok, thanks for the update