WebODM/WebODM #1618: Error processing dataset

aeciolemos, 2025-02-26:
How did you install WebODM (docker, installer, etc.)?
INSTALLER 2.7.0

What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Chrome 132 on Windows 11
i7 8th gen with 32GB RAM and a 1TB SSD

What is the problem?
I have been trying to process a task with 1750 photos and it has been failing.
I see these errors in the log and it says to report, but I am not sure they are relevant to the error

"python: ERROR Uncaught error! This is potentially bad. Please report it to [http://github.com/OpenDroneMap/WebODM/issues](https://github.com/OpenDroneMap/WebODM/issues): [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\WebODM\\resources\\app\\data\\media\\project/2/task/e9a30f95-2665-43d1-b700-8112e3787b0c/assets\\odm_orthophoto\\odm_orthophoto.tif' Traceback (most recent call last):\r\n File "C:\WebODM\resources\app\apps\WebODM\worker\tasks.py", line 143, in process_task\r\n task.process()\r\n File "C:\WebODM\resources\app\apps\WebODM\app\models\task.py", line 797, in process\r\n self.delete()\r\n File "C:\WebODM\resources\app\apps\WebODM\app\models\task.py", line 1128, in delete\r\n shutil.rmtree(directory_to_delete)\r\n File "C:\WebODM\resources\app\apps\grass78\Python39\lib\shutil.py", line 740, in rmtree\r\n return _rmtree_unsafe(path, onerror)\r\n File "C:\WebODM\resources\app\apps\grass78\Python39\lib\shutil.py", line 613, in _rmtree_unsafe\r\n _rmtree_unsafe(fullname, onerror)\r\n File "C:\WebODM\resources\app\apps\grass78\Python39\lib\shutil.py", line 613, in _rmtree_unsafe\r\n _rmtree_unsafe(fullname, onerror)\r\n File "C:\WebODM\resources\app\apps\grass78\Python39\lib\shutil.py", line 618, in _rmtree_unsafe\r\n onerror(os.unlink, fullname, sys.exc_info())\r\n File "C:\WebODM\resources\app\apps\grass78\Python39\lib\shutil.py", line 616, in _rmtree_unsafe\r\n os.unlink(fullname)\r\nPermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\WebODM\\resources\\app\\data\\media\\project/2/task/e9a30f95-2665-43d1-b700-8112e3787b0c/assets\\odm_orthophoto\\odm_orthophoto.tif'\r\n\r\n"

How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app/)
I set up a Fast Orthophoto task with 1750 photos. I set it to run. It runs for about 6.5h then fails. I don't know where to look for errors, I have attached the app log. I am running the same task again now.

[app.log](https://github.com/user-attachments/files/18981834/app.log)
[pgsql.log](https://github.com/user-attachments/files/18981835/pgsql.log)

[image]

aeciolemos, 2025-02-26:
One more clue, it is stuck on Calculating face qualities 72%. It has been at that point for about 1h now.

I can provide copy of the images, however, it is 1750 12MP images, so it will be a huge dataset to copy to GDrive or Dropbox or something and it is client information, so I would probably need a property release form from them.

[image]

pierotofy, 2025-03-12:
> The process cannot access the file because it is being used by another process

Something else (AV/security software?) is causing this error; there's nothing we can do if another process is accessing the files WebODM needs to process.

Glad you figured out the other issue to be caused by low memory.

closed 2025-03-12 by pierotofy