WebODM/WebODM #1637: Fragmented Point Cloud Issue in WebODM on VPS

kemaluzel, 2025-03-27:
I have a dataset consisting of 240 images. When processing it on a VPS ( WebODM 2.7.2) with 4 CPUs and 16GB RAM using the following parameters: “auto-boundary:true, rolling-shutter:true, dem-resolution:0.4, dsm:true, feature-quality:high, optimize-disk-space:true, max-concurrency:2, tiles:true, gps-accuracy:0.1, no-gpu:true, orthophoto-resolution:0.2, use-exif:true”, the point cloud appears fragmented.

[image]

However, when I run the same process on Desktop WebODM without “no-gpu”, the point cloud is clean. ( WebODM 2.7.2)

[image]

If I enable “no-gpu” on Desktop WebODM, the point cloud also becomes fragmented, just like on the VPS. The main issue is that even if I remove “no-gpu” or “max-concurrency:2” on the VPS WebODM, the point cloud still gets generated in a fragmented manner. I would greatly appreciate any help in resolving this issue.

[https://wetransfer.com/downloads/16b7404a99c5cb0973ad7e60f84eadf720250326105554/b3eff5?t_exp=1743245754&t_lsid=5e742a2d-9a87-4115-b73d-7d6c009bf4f4&t_network=link&t_rid=ZW1haWx8Njc4YWFmNTY0ZGZkZTcwNmYwM2QzNDcw&t_s=download_link&t_ts=1742986570](url)

https://wetransfer.com/downloads/16b7404a99c5cb0973ad7e60f84eadf720250326105554/b3eff5?t_exp=1743245754&t_lsid=5e742a2d-9a87-4115-b73d-7d6c009bf4f4&t_network=link&t_rid=ZW1haWx8Njc4YWFmNTY0ZGZkZTcwNmYwM2QzNDcw&t_s=download_link&t_ts=1742986570

kemaluzel, 2025-04-11:
I performed a clean installation of WebODM 2.8.0 on Ubuntu Server 22.04, but the same issue (fragmented point cloud) persists. Could you please help me resolve this problem?

harunalt, 2025-05-12:
I'm experiencing the exact same issue described here. When I run WebODM on a cloud server, this problem occurs. Could the developer team please help resolve this issue?

pierotofy, 2025-09-09:
This is due to GPU vs. non-GPU runs on ODM. Moving to the ODM repo.