WebODM/WebODM #1671: (Improvement) See all orthophotos(models) in project at the same time

Ruthalas, 2025-05-12:
When I take take photography of a neighborhood, I tend to chunk it up into batches of roughly 300 photos to make sure I can consistently process the data.
This results in several batches within a single project.

It would be amazing to see both the orthophotos and the models simultaneously to spot gaps in the overlap prior to uploading to OpenAerialMaps (for example).

I suspect rendering the mode/pointcloud would be more challenging, but even just seeing all the orthophotos in the project at the same time would be a wonderful improvemtn.

Saijin-Naib, 2025-05-12:
You can see all the Orthos at once provided you have all the Tasks in one Project and click the View Map button on the top right of the containing Project.

Ruthalas, 2025-05-13:
Thank you, that is exactly what I was looking for!

Saijin-Naib, 2025-05-13:
As you stated, combined mesh/cloud view is not so simple, and I'm not sure the experience would match expectations if it were simple as one could very easily exceed max number of points/texture dimensions/verts and cause visibility culling, browser/driver crash, etc.

Such a composite might need aggressive LOD to stay within a reasonable resource consumption envelope.

arpitpara, 2025-05-19:
I have tried the View Map options in the latest release and is one of the best feature. I have been using WebODM for construction monitoring from around 4+ years now. 
The only problem I come across is, if I have too many tasks in the same project.(500+)
1. It takes lot of time to load the combine view Map option. 
2. If you can provide 'Select All' or 'Deselect All' Option as I only want to compare 2 orthophotos at a time mostly. I have to go on deselect all 498 orthos in the project.
3. I have seen the merge point cloud option in Potree, they provide check boxes to showcase which point cloud to show.(I understand that volume of points is very large, but can be provided with a limit of 4-5 point clouds).
4. If we can merge orthophotos based on selected tasks and export them as one, can be great feature to add.

Springreens, 2026-11-14:
#4 If we can merge orthophotos based on selected tasks and export them as one, can be great feature to add

Yes... lets say 2 submodels have alread gone thru the expensive matching stage lets say 300 image and a second submodel the same... to be able to join these two submodels by utilizing the work that has alread been done. maybe similar to the align option.  maybe by combining the output files into a new model?