WebODM/WebODM #1700: DroneDB share button state

ozgurozc, 2025-07-08:
[image]

When you want to share with DroneDB and something goes wrong the share buttons hangs up in shareing mode.
I think it should allow to retry . This state does not change even you restart the server, or make an update.

[image]

pierotofy, 2025-11-11:
Update on this one @HeDo88TH ? Is there interest in improving this?

HeDo88TH, 2025-11-11:
Yes, I will take care of this as soon as possible

pierotofy, 2026-01-28:
Should be addressed with #1801 

Feel free to re-open if the issue persists? 🙏