WebODM/WebODM #1705: Relato de Erro - Discrepância entre medidas no modelo 3D e na ortofoto (2D)

luisgvfreire, 2025-07-16:
Hello,

I would like to report a significant discrepancy between the measurements taken from the 3D model and those from the orthophoto (2D) generated by the software.

I understand that terrain elevation can influence measurements, and that measurements based on the orthophoto are naturally limited to two dimensions (2D). However, the difference observed between the 3D and 2D measurements is extremely large—well beyond what would be expected from topographic variation alone.

This discrepancy is compromising the reliability of the measurements for both technical and practical purposes.

I would like to know if this is a known behavior of the software, if there are any recommended best practices to avoid this type of discrepancy, or if it could be the result of a processing error.

Below are screenshots clearly showing the stark difference between the measurements.

[image]

Measurements from the orthophoto (2D)

[image]

Measurements from the 3D model

If needed, I can provide additional information, files, or any other data that may help investigate the issue. I’m available to assist.

pierotofy, 2025-07-16:
Don't use the 3D model for measuring volumes, the algorithm from Potree for measuring volumes gives you the volume of the bounding box of the points (if I remember correctly), which is probably not what you want.

This could be improved in the future.

luisgvfreire, 2025-07-16:
> Não use o modelo 3D para medir volumes, o algoritmo de Potree para medir volumes fornece o volume da caixa delimitadora dos pontos (se bem me lembro), o que provavelmente não é o que você quer.
> 
> Isso pode ser melhorado no futuro.

Thank you for the response.

Just to clarify, I’m not trying to measure volume—what I need is to measure an area that follows the topography of the terrain, not a flat (2D) projection.

The 2D orthophoto gives me a planar area, which underestimates the true surface area when there’s significant elevation change. That’s why I tried measuring it on the 3D model.

Is there any recommended way to calculate surface area that considers the terrain in WebODM?