WebODM/WebODM #1720: Consider using unique Docker container names

Christopher-Hayes, 2025-08-05:
## Description

Right now, the container for the database uses a generic name, "db". For my setup this is causing a collision with another application that also uses "db" for the database.

Consider using a container name like `webodm-db` to avoid Docker container naming collisions.

This could be standardized across the Docker containers:

- `webodm-broker`
- `webodm-webapp`
- `webodm-db`
- `webodm-worker`

Christopher-Hayes, 2025-08-07:
> How do we migrate exisiting setups without breakage?

Yeah, that would be a challenge.

Saijin-Naib, 2025-08-07:
Yeah, and that will need to be addressed before we could roll out such a change. 

Who all does it impact negatively, and how do we mitigate that?

pierotofy, 2025-08-07:
I'm inclined to suggest that in case of collisions, one can just edit the docker-compose files and choose a name that fits their setup.