WebODM/WebODM #1722: LAI equation

cvandenbergTrace, 2025-08-07:
Plant Health | LAI equation appears incorrect. 
app/api/formulas.py

Current equation reads as:
'3.618 * (2.5 * (N - R) / (N + 6*R - 7.5*B + 1)) * 0.118'

Several online sources show the following equation.

**LAI = (3.618 × EVI) - 0.118**
Where EVI (Enhanced Vegetation Index) = 2.5 × (NIR - Red) ÷ (1 + NIR + (6 × Red) - (7.5 × [Blue)

[Link 1](https://geonetwork.swissdatacube.org/geonetwork/srv/api/records/4f68aa2b-47a0-407d-a289-bf943e669cb9)
[Link 2](https://catalyst.earth/catalyst-system-files/professional-help/concepts/focus_c/oa_featExtract_intro_vegIndFeat.html)
[Link 3](https://medium.com/geoinfomatics/retrieving-leaf-area-index-lai-google-earth-engine-gee-6b7fadab9291)

kikislater, 2025-08-07:
But formula you cite needs Nir band whereas there are some studies with LAI estimation from RGB ...
i.e: http://dx.doi.org/10.3390/su11236829

So may be it is needed to retrieve history for this Formula in odm ... And may be the purpose was to use LAI with RGB drone

cvandenbergTrace, 2025-08-07:
> But formula you cite needs Nir band whereas there are some studies with LAI estimation from RGB ... i.e: http://dx.doi.org/10.3390/su11236829
> 
> So may be it is needed to retrieve history for this Formula in odm ... And may be the purpose was to use LAI with RGB drone

The webodm equation also needs the NIR band.
Current equation reads as:
'3.618 * (2.5 * (N - R) / (N + 6R - 7.5B + 1)) * 0.118'

Here, 3.618 and 0.118 are also both multiplied across the same part of the formula. It is equivalent to:
0.0426924 * (2.5 * (N - R) / (N + 6R - 7.5B + 1))

closed 2025-08-07 by pierotofy