WebODM/WebODM #1728: Accessibility: Remove/Redesign Confetti Effect On About Page

Saijin-Naib, 2025-08-19:
A user with a medical condition affecting their vision brought up that this effect is not only distracting, but distressing, as it appeared to them to be symptomatic of their condition reoccurring.

Could we consider removing or changing this effect?

pierotofy, 2025-08-19:
Today I learned that a detached retina can cause eye floaters.

Saijin-Naib, 2025-08-19:
I always thought this page was a cute reward for a successful upgrade, but this is why enabling people to share and learning from others is so important.

I never would have thought this would be an issue for someone, because it isn't for me, so it wasn't on my radar, what-so-ever.

Thanks for the quick fix!

PR #1729 (2025-08-19): Remove stars effect
Closes #1728