WebODM/WebODM #1735: nvidia gpu node not working.

Bkoltz, 2025-08-28:
Hi,
Not sure if this is user error or not, but I am running WebODM just fine on my Docker container, and have a separate container dedicated to running the GPU Node for my setup. I am running TrueNas Scale v25.04.2.1 with a NVIDIA GT 1030 GPU. Recently (couldn't tell when exactly) I haven't been able to strat my GPU node container and the only error I get is "timed out, took to long to respond." If anyone has input on this that would be great. Here is my yml for the GPU node container:

services:
  node-odm:
    deploy:
      resources:
        reservations:
          devices:
            - capabilities:
                - gpu
              count: 1
              driver: nvidia
    expose:
      - '3000'
    image: opendronemap/nodeodm:gpu
    oom_score_adj: 500
    ports:
      - '3004:3000'
    restart: unless-stopped
version: '2.1'

pierotofy, 2025-08-28:
Mm, it doesn't look like a bug. Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

closed 2025-08-28 by pierotofy

Bkoltz, 2025-08-28:
@pierotofy Yes, I didn't realize that there was a forum. Thanks for the info!