WebODM/WebODM #1747: NB 2 Urgent Issues - Some file not found plus dataset could not be completed

booysenherman25-cpu, 2025-09-20:
Hi, 
The last 2 weeks, I have been having significant issues with the program.
I've attached a screenshot for you to look over. This has happened 4 times with two different mapping sites.
Concerning the screenshot, it was only 624 photos where 2 of the images were on a different memory card, so the error database could not be completed 1 hour and 30 minutes later, after a message Queued (something I keep on seeing regularly lately). I had to restart Web ODM and then this attached code came up.
Before I could upload more then 2000 photos and now the same problem with 624.
Why is there no error message when one uploads the photos, only to have the error message appear a couple of hours later?
I have been using Web ODM for more than a year with no issues. Over the last two weeks, I have had to delete and restart projects multiple times

[image]

The first time it happened was after a 22-hour 45-minute dataset error.

Please help

Kind regards
Herman

booysenherman25-cpu, 2025-09-20:
Hi, with possible software fault what to do next?

Could you send me a new instalation file, I have Windows 11 installed on my laptop.

Hope to hear from you soon as I like Web ODM...easy to use.

The drone I use is and DJI Air2S

How many photos can be prossessed in one upload?

Kind regards.

pierotofy, 2025-09-20:
Hi, this actually looks like a question. Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1: